BlocklyDuino/BlocklyDuino #17: Willing to help, what are the priorities?

rmetrich, 2015-02-12:
Hello,
I'm willing to help develop this software which seems a good way to learning programming specially on Arduino and seems more effective than Arduiblock.
I just forked the code and would like to help.
My priorities are:
1. translate to french (in order for my kid to be able to program)
2. be able to do basic programming of his DFrobot's insect robot, requiring currently missing blocks such as math stuff (e.g. random number)

What are your priorities?

gasolin, 2015-02-13:
Thanks for willing to help :) 

The first priority is need more eyes to validate this patch
https://github.com/gasolin/BlocklyDuino/pull/16

It update BlocklyDuino's blockly core up to date and bring localization into BlocklyDuino. From BlocklyDuino's point of view we need make sure the generate code still works fine with this patch.

it will be nice If you could help check it.

rmetrich, 2015-02-13:
OK, will have a look.
For now, I have some issues compiling the code: seems the build.py only creates a compressed version of the 'arduino' javascript code, which is problematic to me as it seems to use an API on the internet and I have issues with my internet connection.

gasolin, 2015-03-02:
Hi new version is available with latest blockly core. Have you fixed the compile problem?

SebCanet, 2015-03-03:
About translation in French, I've done it : https://github.com/technologiescollege/BlocklyDuino
Don't waste your time doing it again, there're much funny things to do to enhance it :-D !

rmetrich, 2015-03-03:
Sorry, I lacked time these weeks.
I did apply the initial patch then tried to merge it in the current branch with some difficulties.