BlocklyDuino/BlocklyDuino #18: New blocks, block fixes

stevoleeto, 2015-02-13:
I've modified my fork to a certain extent over the past few months

I've added new LCD blocks to work with most LCD's that come in a kit.
I've added blocks to be used with Sparkfun's 9 Degrees of Freedom Sensor Stick (Gyro, Compass, and Accelerometer).
I'd added blocks to be used with a shift register.

I've also fixed a few bugs throughout my time looking at it, specifically with previous blocks not working as expected etc. 

I'm currently running it on a school server here, 
http://acsweb.ucsd.edu/~sgilardi/blockly/apps/blocklyduino/

gasolin, 2015-03-02:
@stevoleeto If possible please fire issues with separate PR of each independent changes. So others could get benefit from your contribution

stevoleeto, 2015-06-27:
@gasolin  Will do! Sorry this has taken so long. I will be doing this in the next few days.