BlocklyDuino/BlocklyDuino #20: check howto make BlocklyDuino internationalizable

gasolin, 2015-03-02:
Since latest merged blockly core has internationalization architecture, BlocklyDuino should align to blockly and provide strings in that way.

SebCanet, 2015-03-03:
Thanks a lot !
A spanish teacher asked to translate my version, but he has to rewrite everything...
I hope with this new core it'll be more easy.
Thanks !

SebCanet, 2015-03-10:
FINISHED ! I rewrite a totally new BlocklyDuino from Blockly's example 'code' with language menu.
Thanks a lot for your work taht helped me to start this work that is totally new from scratch. But as I use some of your code for Grove module, I mentioned your work. But I have to rewrite everything in Grove module to use 'relative' text from language file.

gasolin, 2015-04-02:
great, would you like to point out your repo so somebody can redirect to your repo from here, or make a PR for BlocklyDuino?

commit 178c3481 (2016-09-01): fixes #20: always use latest workspace to upload code to board
commit 8cdb83b5 (2016-09-01): Merge pull request #81 from Smartrobotics/gh-pages

fixes #20: always use latest workspace to upload code to board
closed 2016-09-01 by gasolin