BlocklyDuino/BlocklyDuino #29: Few issues

rsrini7, 2015-05-11:
I have tried in Ubuntu 14.10. Following are the observations.

First error:
Uncaught TypeError: Cannot read property 'value' of null

So, changed in function uploadClick() from blockly_helper.js
'textarea_arduino' to 'content_arduino'

in Port 8080
Arduino INO web server
To upload to an Arduino board connected to this computer, POST to /.

In Port 8000
Application Cache Error event: Resource fetch failed (404) http://127.0.0.1:8000/apps/blocklyduino/BlobBuilder.min.js (But i ignored)

XMLHttpRequest cannot load http://127.0.0.1:8080/. No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'http://127.0.0.1:8000' is therefore not allowed access. The response had HTTP status code 501.

SimpleHTTPServer started with below header. Now the above issue gone.
self.send_header('Access-Control-Allow-Origin', '*')

Next:
POST http://127.0.0.1:8080/ net::ERR_EMPTY_RESPONSE (Chrome Console)

Terminal
mkdir: cannot create directory ‘ino_project’: File exists
sh: 1: ino: not found
ino init returned 32512
127.0.0.1 - - [12/May/2015 00:53:26] "POST / HTTP/1.1" 501 -

What is next ? Need help. Is this the way ? Can you please give me the steps ?

rsrini7, 2015-05-31:
Can anyone help me please? Tool is looking good and want to learn and explore.

gasolin, 2015-06-16:
Hi, thanks for step and fixing issues. 

I think you need to install modified version of ino tool. But I think you may like try to implement #32 if you know python well.

gasolin, 2015-06-16:
`ino: not found` means you did not install ino tool. check https://github.com/gasolin/BlocklyDuino#integrated-arduino-upload to install ino tool

gasolin, 2015-06-16:
update above addressed issues, thanks!

closed 2015-06-16 by gasolin

rsrini7, 2015-06-23:
Thanks. i tried to install inotool and it is not supporting the latest arduino then i have installed "http://32bits.io/Arturo/" which works well. But the problem is now i am facing same origin policy between blockly web and the arduino listener. I tried with adding in Header set Access-Control-Allow-Origin "*". Still brower is blocking. Any suggestion ?

mateusdepaula22, 2017-09-03:
Hello, I have a problem, could someone help me? I use raspberry pi as a BlocklyDuino server computer. When I try to access from a browser on another computer on my network when uploading, identification errors occur. How do I solve this?