BlocklyDuino/BlocklyDuino #41: Co-create blocks

SebCanet, 2015-07-04:
Hi,
there's a lot of forks of your wonderful idea. But I think we must not work separatly, waste time.
At this time, your co-helper does not create blocks the right way, and the internationalisation problem.

I think we should work to create "good way" to create blocks, I started on my wiki : http://tic.technologiescollege.fr/wiki/doku.php/en/blockly_rduino/create_blocks
but I don't know if everything's all right...

We should work together on blocks, creating a blocks library, taht anyone could download. And because everything started with your work, you could create a repository specific for blocks.

Hope we could all work together, helping everyone use this great tool in an 'universal' way.
I'm ok to help you in any way with that.

Regards.

gasolin, 2015-07-06:
It's a good idea to have reusable blocks. Though currently we did not have much way to share them in proper format.

I was thinking to make a blockfactory-like tool to create and submit the arduino blocks more easier, but still lack of time :/
https://blockly-demo.appspot.com/static/demos/blockfactory/index.html

stevoleeto, 2015-07-06:
@gasolin  We could get started on a version of this. Here's an example of something I just did with the code factory. We could expand on it more to get it to create the generator code.

[image]

SebCanet, 2015-07-06:
Hi,
I sent mails to TC teacher to ask if this could be an exercise for student....because I have no time for that and no special knowledge in it....

I think first we should prepare a 'good' way to create blocks, as BitBloq : https://github.com/bq/roboblocks

For tests : https://github.com/technologiescollege/Blocks-Arduino

What do you think of that ?

stevoleeto, 2015-07-07:
Agreed. In the meantime, I'll contiune to mess around with the Block Factory stuff and see what I can come up with.

carlosperate, 2015-07-07:
Hi all,

This looks quite promising. I have a fork of blockly myself that I originally started developing about a year ago when it looked like Bloklyduino wasn't being actively maintained and there were tons of blockly updates that I personally needed for my project (together with static typing). Now it looks like this is gaining a bit of traction again and it'd be exiting to see it grow.

@gasolin Since you mentioned that you haven't got that much time to work on this, I was wondering if perhaps it'd be a good idea to create a Blocklyduino github organisation to create a more community-driven set of repositories? We could have the following:

* Blocklyduino
* The "blocks" repository discussed here
* A "block creator" project based on blockly factory
* Perhaps maintain a forked version of blockly with the blocklyduino updates to easily pull from upstream and then just keep the compressed js libraries in the Blocklyduino repo)

I don't know, what do you guys think?

SebCanet, 2015-07-07:
I think it's a really damn good idea !!!!
Let's do it !

gasolin, 2015-07-08:
@carlosperate thanks for the further proposal.

I have some concerns about moving to an github organization. 

The very first is forming the organization wont gain organizers automatically. 
We need have several active contributors first then consider forming the organization.

2nd is forming the organization wont make projects active automatically. The dreamed block creator or blocks repository wont be happen until someone take the mission on his shoulder.

3nd is about review process, its necessary because there always should be someone responsible for the code quality. And in organization everyone has permission to merge code. 

But I have no strong objection about this idea, its ok if block creator or blocks repository are there and some people love us, and then we should put them together to make people feel life easier.

carlosperate, 2015-07-08:
Hi @gasolin,

I completely understand, and you are certainly right in that making an organisation won't automatically bring collaborators. 

The reason I suggested it is because if we created repositories related to Blockyduino, but are not somehow officially promoted by Blocklyduino itself, they won't get any traction either.

When I created my own blockly fork for arduino I did it at a time when Blocklyduino was terribly outdated and stagnated in terms of new features (not significant update in much over a year and pull requests took months to even get reviewed). So I set off to work on my own version based on blockly master as I though it would be less hassle than trying to update Blocklyduino itself (I was proven wrong by Neil's contribution). That is not the case any more, and I must admit I could have tried to contribute here instead of keep working on my own fork, but I had my reservations about the amount of support given, taking in consideration that even Neil's (Blockly's main developer) pull request took a month to get accepted.

Please don't get me wrong, I'm not trying to criticise, everybody has limited time and you are doing this for free and in the open, for which we are all quite grateful. I guess what we need is more contributors, but it might put some people off if even the simplest of pull request might takes ages to get reviewed.

I'm not quite sure how much time you can devote on Blocklyduino, or how much time you'd like it to evolve (how do you feel about staying up to date with blockly master?). Of course it is your project and you should have a final say on what is part of it or not, maybe you could use the help of collaborators? I'm not quite sure how you feel about that.

>2nd is forming the organization wont make projects active automatically. The dreamed block creator or blocks repository wont be happen until someone take the mission on his shoulder.

I would be committed to get blockly factory for arduino blocks up and running, specially taking in consideration the new changes coming to the upstream blocks construct and blockly factory, although it seems like @stevoleeto already had a start. 

But how would additions like this be combined with Blocklyduino? A different repository to be maintained separately? A fork with to be merged back into this repo?

I am also very interested to get this idea of a collection of blocks that are easy to import, but I do fear that if it is a repo maintained by somebody else it will just go unnoticed, and this is not something that would fit inside inside this repository.

> 3nd is about review process, its necessary because there always should be someone responsible for the code quality. And in organization everyone has permission to merge code.

I completely agree, there should be a gatekeeper to ensure code quality, I assumed there could be different privileges given to different repositories, but if every member has full access to everything that's a bit more dangerous. I suggested it because I though it would be a good thing to have something that captures all repositories related to a project, instead of having them in a user repository list with a bunch on unrelated projects. I guess you could have a "one person" organisation to group repositories, but there is no point until there are repositories to group.

carlosperate, 2015-07-17:
Is anybody still interested in the original topic of the block factory and block collection? @stevoleeto where you able to advance the preview you showed us?

stevoleeto, 2015-07-18:
@carlosperate Depends on how far we want to go with it. I tinkered with it a bit, but it's on par with what a typical block factory does. It generates the block definition and just has the user finish the Arduino definition for the block.

carlosperate, 2015-07-18:
Well, that's still progress. Which version of the block factory have you used? The latest one will need Blocklyduino to be updated to the latest blockly.

I'm thinking having the main blockly area on the left divided on 2 workspaces (that will also need the latest version of blockly), the top one stays as it is with the factory blocks, and the bottom one would have all the blocks similarly to the "blocklyduino" app in the app folder. Then we can use the blocks to generate the "generator code" as well (of course it will need the creation of special blocks to input the data from the block input fields, and to be able to insert calls to new functions/libraries).

I'll see if I can work on a quick draft and show you what I mean.

carlosperate, 2015-07-18:
This is more or less what I had in mind:

[image]

Nothing is really implemented, just a quick play with the layout.

stevoleeto, 2015-07-19:
Sweet, let's get working on this! Working on BlockFactory may also be something to contribute to Blockly in general as well. Where should I upload what I have started so we can continue working on it?

carlosperate, 2015-07-19:
If you add it to your Blocklyduino fork it can then be updated there or pull requested into any other fork.

I have just submited a pull request to update Blocklyduino to the latest blockly, while it gets reviewed we could continue on my working updated version (https://github.com/carlosperate/BlocklyDuino).

If you add it to your fork first, make sure it is inside the app folder.

SebCanet, 2015-07-19:
I don't know if I did it well, but I created an organization to try to work together.
https://github.com/Blockly-ar-Duino
What do you think of that ?
I give admin rights to everybody to start, because I'm not very ease with Github....

Hope it could help.

carlosperate, 2015-07-20:
The way I look at this, it can only reach decent exposure if the block factory and collection repositories get somehow adopted by Blocklyduino (either pulled into the Blocklyduino repository, or as an organisation with individual repos as previously discussed).

I don't mind having a different github organisation while we work on this, but if the end goal is to merge it back or transfer the repositories somewhere else then it would be more of a temporary space more than anything else, which could could get by using forks.

Also, I think that the repository for the blocks collection (and the factory) should have a reference to "Blocklyduino" in the name somehow (Blocklyduino-blocks would be the most straight forward, yet unoriginal, title). On this note, it also does not make that much sense yet to create a repository without agreeing on a format for the blocks, which in my opinion is the most important aspect of the collection part of the project. Once a format is agreed upon any change could potentially break any previous implementation by anybody else. We could possibly use the github wikis/issue tracker for this (to keep a new repository history clean), and once again this could be done in a fork. 

This is all, of course, if we have the blessing from @gasolin . As it is his project and his project name, if he feels that this is a direction he would not like to take with Blocklyduino then we should respect that and if we are still interested to pursue this we should do it under a different name.

gasolin, 2015-07-22:
Thanks for all your passion to make BlocklyDuino better than ever. 

After some research I found github able to give different permissions to each repository in an org. Therefore there's no resistance to migrate BlocklyDuino there. I created the org at https://github.com/BlocklyDuino and assign full admin permissions for BlocklyDuinoFactory and blockslibrary to you, so you could do prototyping in full speed. 

As above comment, I'd still conserved to provide the merge permission for BlocklyDuino itself, hopefully we do able to attract more contributors and figure out a working model.

carlosperate, 2015-07-22:
Thank you @gasolin! This should really help us get this up and running quickly with more control. And thank you also @SebCanet and @stevoleeto for the great idea and starting all of this, the possibilities are quite exiting :)

SebCanet, 2015-07-22:
Thanks a lot !!!!
Lets' go creating !

stevoleeto, 2015-07-25:
@carlosperate @SebCanet @gasolin I went ahead and pushed the files I was using into the BlocklyFactory repo.

carlosperate, 2015-07-25:
That's great! Unless you are planning to still work on it tonight, tomorrow I'll update it to the latest Blockly/BlocklyDuino to add a second workspace.

stevoleeto, 2015-07-25:
@carlosperate Nope! Go ahead.