BlocklyDuino/BlocklyDuino #48: migrate project to https://github.com/orgs/BlocklyDuino/

gasolin, 2015-07-22:
as discussion in #41, create an organization to host BlocklyDuino related projects

gasolin, 2015-07-22:
migrated to https://github.com/BlocklyDuino/BlocklyDuino

closed 2015-07-22 by gasolin

carlosperate, 2015-08-13:
Is the gh-pages branch the new "master" branch?

gasolin, 2015-08-14:
yes, therefore we'd have always update version

carlosperate, 2015-08-14:
I wonder how this is going to affect forks relaying on "git pull remote
master" to update their version with the latest upstream. Maybe it'd be
better to delete the master branch to produce so that it produces an error
in pull and forces the users to check the repository in github?

gasolin, 2015-08-16:
Sorry for not notice the merge requirement. How about try with rebase than merge?

```
$ git remote update
$ git rebase upstream/gh-pages
```

carlosperate, 2015-08-17:
I don't think there is any real issues rebasing or merging the contents of the branch. What I was trying to communicate is that I would assume most forks might try to pull the latest changes by doing a git pull from master. So, unless they happen to follow the issue tracker, or randomly notice that the main branch is not "master", but "gh-pages" (very easy to miss on the github interface), then most people following their usual work-flow will not be getting the latest changes, as a `git pull upstream master` will still sync with an old version of BlocklyDuino. 

(also, I just realised I asked about this in the wrong issue, meant to comment on https://github.com/BlocklyDuino/BlocklyDuino/issues/53 )