BlocklyDuino/BlocklyDuino #49: grove hw images

carlosperate, 2015-07-23:
I was wondering if there was an specific reason why the grove block images were being hotlinked. If we copy the images into the repository (with the appropriate attribution), they would be available offline and we could also remove the background:

[image]

gasolin, 2015-07-23:
No particular reason, Just feel lazy. 
For people who don't want to use grove modules they could remove blocks directly without concern images.

Would image storage also be considered into shareable blocks? #41

carlosperate, 2015-07-23:
> Would image storage also be considered into shareable blocks? #41

That's a very good point, we'll need to look into that. I cannot think of a simple way to do that in javascript only.

gasolin, 2015-07-29:
maybe base64 images can serve it well

carlosperate, 2015-07-29:
Is that what the text block uses for the quotation marks?

gasolin, 2015-07-29:
you could try with some online tool such as http://www.base64-image.de/

Arachnid, 2015-08-13:
I think it would be a good idea to remove the images entirely. I don't believe they add anything to the usability, and they certainly detract from the visual appeal.

gasolin, 2015-08-14:
Really? In contrast I even want to put every logic blocks with images so child can learn programming before they can read words :p

Arachnid, 2015-08-14:
I think that's a noble sentiment, but I don't think it's very practical. I'm pretty sure a child too young to read won't be able to make the leap from the Grove sensor in their hand to the picture onscreen, much less know how to use it!

Arachnid, 2015-08-14:
I was talking to the teacher I'm working with, who's now tried both versions, and she thinks the version without images are easier for kids to work with.

If you disagree, though, would you accept a patch that enabled or disabled images in blocks globally with a flag, which was set to on by default? That would make it easy for maintainers of forks like me to turn it off if it's not wanted.

SebCanet, 2015-08-14:
Hi,
I'm back. About images, in France, every teacher or educator I'm working with think it's useful...sorry.
You already have the option collapse/expand block, is it what you want ?

Arachnid, 2015-08-14:
Collapsing the block hides useful information about selected parameters; what I was talking about is saving space by omitting the image entirely. I think providing this as a compile-time option would be the best of both worlds, personally, and make it easier to maintain a branch without them.

carlosperate, 2015-08-17:
> I think providing this as a compile-time option would be the best of both worlds

Not quite sure how you were thinking about implementing that, as it seems like an unnecessary addition to the build file. However, I do agree that having a flag that enables or disables the images on run time would not be a bad option.

And about:
>I'm pretty sure a child too young to read won't be able to make the leap from the Grove sensor in their hand to the picture onscreen, much less know how to use it!

You'd be surprised, a lot of kids these days learn how to use tablets and computers before they learn how to read. I've seen 6 year olds writing actual code (in a workshop designed for older kids) and 11 year olds struggling to use scratch.

Arachnid, 2015-08-17:
Regarding enabling it, I was figuring on setting it as a parameter in one of the JS files that the blocks depend on, and checking it in each block. Then enabling or disabling it is a oneliner.

Were the 6 year olds able to read? I'm struggling to see how someone who couldn't read would find these images useful: they might be able to identify the block, but how would they know how to configure it? And other blocks, like 'if' and 'digitalRead' and 'loop' don't come with pictures.

gasolin, 2015-08-18:
I'm fine if someone want patch the image block to make it can be configured to not shown on screen.
If it's there we can add images to `digitalRead` / `loop` blocks and see if it helps for younger children :-)