BlocklyDuino/BlocklyDuino #58: Consider removing no-op grove blocks

Arachnid, 2015-08-13:
Some of the grove blocks, such as the ultrasonic rangefinder, clearly add a lot of value, because they provide an interface to library functionality. A lot of them, though, such as the button and buzzer blocks, are exactly identical to standard blocks like digitalRead, digitalWrite, and analogRead.

Blockly's philosophy is to make the blocks visually resemble the code that's generated in order to make transitioning from visual to textual programming as easy as possible. I don't believe we're doing students a service by hiding what they're actually doing when interacting with these blocks, and it negatively impacts the readability of the blocks vs the code.

Finally, if a student wants to use one of these devices with a different function (such as a buzzer with the tone block), they need to learn how to treat them as regular devices anyway.

I think it's worth considering removing any blocks that replicate standard functionality like this.

carlosperate, 2015-08-14:
I disagree, I think specially for somebody starting out fresh, they can initially use the grove blocks (as it is a very easy way to identify what they are using) and then they have the option to learn how to use the normal IO blocks for their control if they wish to do so.

stevoleeto, 2015-08-14:
I actually agree @Arachnid on this one. Less so about their functionality, but rather how they focus on one specific kit. We could choose to support multiple beginner kits (Arduino, Sparkfun Inventor, etc etc), and have specific kits chosen at compilation if desired.

SebCanet, 2015-08-14:
I used Bockly with 9 years kids, after Scratch initiation. It worked well. But I think we need specific blocks to help newbies, like kids.
For multiple support, I did it on my version : https://github.com/technologiescollege/Blockly-at-rduino
If you want test and take what you need, what you want to enhance.
 I'll work on what we need in blocks this week-end for the repo.

Arachnid, 2015-08-14:
@stevoleeto I think that being able to select which categories get included dynamically would be an excellent feature to have.

gasolin, 2015-09-10:
For me I like to have visual aid rather than plain digitalRead/digitalWrite, but it's just my taste.
Maybe we could add help string in those module to hint user that they could use normal block instead.

And it will not be an issue if we have a dynamic set to load blocks

carlosperate, 2015-09-10:
If blocks can be dynamically added, and a option to display or hide an image is also included, maybe this post from the Blockly user group can be useful if the blocks are not correctly re-rendered https://groups.google.com/forum/#!topic/blockly/4BR1trqQ4gw