BlocklyDuino/BlocklyDuino #60: Variables don't do typechecking

Arachnid, 2015-08-14:
While other blocks check types (eg, if, comparison), any block will accept a variable. For instance, you can't do "if(0)", but you can do "if(x)" where x is an int.

carlosperate, 2015-08-15:
With the current implementation it is not possible to discriminate by type of a variable as the blockly implementation itself does not distinguish if the variable has a specific type.

The Arduino generator has defined all variables to int, but that happens only on code generation so you could have a variable equal a pin input and you can use that for an if statement.

gasolin, 2015-09-10:
https://github.com/BlocklyDuino/BlocklyDuino/pull/27 seems fix part of this issue, 

since the origin pull requester does not response a while, welcome picking that and make a fix

carlosperate, 2015-09-10:
Well, https://github.com/BlocklyDuino/BlocklyDuino/pull/27/ only adds a type during code generation, and only by using a value hardcoded into a drop down field. Blockly itself has no clue as to which "type" a block is in the workspace, unless there are additional attributes added to the blocks themselves. This would require to edit every single block definition in Blockly, which can be quite a bit of work (something that I have done myself in my own fork, although currently broken due to an unfinished refactoring of the code generator).

Arachnid, 2015-09-10:
The current blocklyduino has at least some support for variable type - you can't drop a string and an int as two arguments to a math operator, for instance.

carlosperate, 2015-09-10:
They are not proper variable types, they are block input and output types hardcoded into the blocks themselves, which can be set to any javascript string. So when you create a block you can specify its [output type like this](https://github.com/BlocklyDuino/BlocklyDuino/blob/gh-pages/blockly/blocks/math.js#L48), and its [input type like this](https://github.com/BlocklyDuino/BlocklyDuino/blob/gh-pages/blockly/blocks/math.js#L69). However, yes you could add extra functions to Blockly core or the blocks to dynamically change this.

The difficult part is when you have a block like a `variable_get`, its type depends on the input block connected to a different `variable_set` block which has the same variable name selected in its drop down. To keep track of that you need to add event listeners to the workspace to change a newly added attribute to the block (or edit on the fly the block input/output or edit the blockly core to analyse and assign types at a specific moment during the runtime), and extra code in the generator to correctly set the type in the code. That all can be done, it just requires to edit pretty much every single block and a few more code modules. The main reason I was refactoring my fork is because I was encountering issues with type assignment being at the incorrect order during "type analysis" (easy fix would be very inefficient, so was trying to design an alternative), but it shouldn't take much effort to finish it, just haven't had the time.