BlocklyDuino/BlocklyDuino #64: File not found BlobBuilder.min.js and favicon.ico // etc error..

leekj1840, 2016-01-26:
[image]

where file.....

and

upload button click

[image]

why error..?

python version 2.5

I want go home.. ;o;

carlosperate, 2016-01-26:
Python 2 needs to explicitly use unicode, if you have python 3 installed as well (and if the script works on Python 3) that might resolve the issue.

leekj1840, 2016-01-27:
[image]

python 3 install = not launch

So i install python 2.7.11

and ascii(x) utf8(o)

but upload button click = fail error

thanks to your answers

carlosperate, 2016-01-27:
If it is failing at a print statement is definitely still a unicode error, might need more modifying to make it fully compatible under python 2, or updated to make it python 3 compatible.