BlocklyDuino/BlocklyDuino #65: Gathering Related Resource for STEM/Education

gasolin, 2016-02-04:
I occasionally got some request about providing some suggestion of using Blockly + Arduino on Education/STEM/Robots

Please share any related experience here.

gasolin, 2016-02-04:
Though I'm not in educational area, if you want student learn Arduino code afterward, I think BlocklyDuino is still the best way to do so.

Another way is through scratch
http://learn.makeblock.cc/learning-scratch/#Step1-Install-the-mBlock--amp--Arduino-driver

Scratch has better IDE, but it will be easier to extend new block through BlocklyDuino instead of scratch.

ekawahyu, 2016-02-10:
I wrote a simple Arduino library to move the robot forward, backward, turning left/right. The library is basically sending JSON string over a serial port. Any pointer on how I can add new blocks and where to look in your code (if somehow I can copy and modify from there)? Do I need to start by creating new blocks from the block factory first?

stevoleeto, 2016-02-10:
@ekawahyu Every block consists of a block definition (creates the block) and a generator definition (generates the Arduino code from the block).

Look at how the previous blocks were defined in those two areas. Let me know if you need more clarification.

ekawahyu, 2016-02-11:
@stevoleeto I tried to make a similar Control block - Delay function with Block Factory, but my Delay missed something:

My Delay generated from Block Factory:
Blockly.Blocks['base_delay'] = {
  init: function() {
    this.appendValueInput("DELAY_TIME")
    ...
}

BlocklyDuino's Delay:
Blockly.Blocks['base_delay'] = {
  init: function() {
    this.appendValueInput("DELAY_TIME", 'Number')
    ...
}

Unfortunately, there is no way to upload a screenshot of my Block Factory in here. What am I missing so that I don't have 'Number' parameter in my Delay block?

ekawahyu, 2016-02-11:
I built the blocks with a missing parameter 'Number' on base_delay block, but everything seems to work just fine. I can drag and modify the delay time, even the code generated is correct. What is that 'Number' parameter used for?

jwill4, 2016-05-05:
"code for life" website has a blockly tutorial which starts at total beginner and is very slow paced - for primary education.  The best part is it has an introduction to python which allows students to begin writing script.

jwill4, 2016-05-05:
I have used the ZUMO32U4 bots by Pololu in year 10 education.  We start with Blocklyduino running blocks i created specifically for the bots. see my repository for more. https://github.com/jwill4/Zumo32U4Blocks