BlocklyDuino/BlocklyDuino #66: you need to add a True block

amigojapan, 2016-02-07:
I tried to make a while loop,   but when I tried to connect a 1 to the while loop to do while(1) , the block did not allow me to connect it, I guess because it is not a logic block...   and when I looked into the logic blocks I could not find the "True" block, which I know is part of the original blockly... so yeah, I wonder how I can make an eternal loop....

Thanks in advanced!

gasolin, 2016-03-20:
Hmm I haven't use that further. The reason why I did not include true block is the general usage can be done by pre-bundled blocks. I'd prefer a separate do-while loop with pre-bundled true block

Welcome to contribute

carlosperate, 2016-03-28:
@amigojapan have a look at #71 to see how to easily add the block into your own fork if you need it.