BlocklyDuino/BlocklyDuino #67: Upload not working in Linux

ronnqvist, 2016-02-21:
When trying to upload my sketch I get this in Ubuntu Linux 14.4:

created sketch at /tmp/tmp9rV_Ue/tmp9rV_Ue.ino
INFO:root:Couldn't find Arduino command; hoping it's on the path!
INFO:root:Guessing port name as None
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 48329)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 295, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 649, in __init__
    self.handle()
  File "/usr/lib/python2.7/BaseHTTPServer.py", line 340, in handle
    self.handle_one_request()
  File "/usr/lib/python2.7/BaseHTTPServer.py", line 328, in handle_one_request
    method()
  File "arduino_web_server.py", line 133, in do_POST
    print "Uploading with %s" % (" ".join(compile_args))
TypeError: sequence item 3: expected string, NoneType found
----------------------------------------

josx, 2016-08-22:
same here, any clue how to solve it!

carlosperate, 2016-08-25:
The problem seems to be that it fail when trying to guess the port name:
```
INFO:root:Guessing port name as None
```
So it cannot join the `compile_args` list into a parsable string to send to the Arduino IDE.

Try manually setting the port name with the `--port` flag.

If that doesn't work check manually setting all flags, you can have a look at [arduino_web_server.py#L74-L76](https://github.com/BlocklyDuino/BlocklyDuino/blob/b670c35b3dddfdcd7fa1098975e94ad2798e7ad9/arduino_web_server.py#L74-L76) to get a general idea of what it expects.

carlosperate, 2016-08-25:
Jose, could you check if the begging of the console output is the same for you? if any of the other compile flags is `None` it would throw the same exception for a different reason.

josx, 2016-08-25:
@carlosperate the same thing, if i set the port with ---port works ok.
thanks for replying

carlosperate, 2016-08-25:
Glad you got it working !

dedenker, 2016-10-20:
Hi Guys, I also get an error when trying to upload.

> created sketch at /tmp/tmpO8K5fC/tmpO8K5fC.ino
> INFO:root:Couldn't find Arduino command; hoping it's on the path!
> Uploading with arduino --upload --port /dev/ttyUSB0 /tmp/tmpO8K5fC/tmpO8K5fC.ino

Mine seems related to the system call with **arduino** binary, which doesn't accept the INO file, also trying manual:
[image]
This is the error I get via BlocklyDuino and doing it manually.

But I can open and upload the INO file created by BlocklyDuino.
Anybody more has this?

A possible solution I though of is using [INO](http://inotool.org/) command line, instead of Arduino IDE.

carlosperate, 2016-10-24:
Which version of the Arduino IDE are you using? That message should read "Arduino can only..." as shown in https://github.com/arduino/Arduino/blob/31a9029c1d89e394f4f8d6948504127120177756/arduino-core/src/processing/app/i18n/Resources_en.po#L223-L227 since the codebase was migrated to github, so around 2014.

Is it possible you are running Arduino 1.0.x or earlier?

dedenker, 2016-10-26:
No, 1.16 recent [download](https://www.arduino.cc/en/Main/Software).
It is the x64 version, for what it matters.

carlosperate, 2016-10-26:
The thing is that the message screenshot you have provided would only appear on really old versions of the Arduino IDE. Could you check which version of Arduino is launched when executing `arduino` on the command line? My guess is that apart from the latest version you downloaded, you also have an old arduino package (probably installed from your distribution repositories, eg `sudo apt-get install arduino`), and this tools is picking the older (unsupported) version instead.

dedenker, 2016-10-27:
I understand, but I just checked again (proper). Same message and my version is 1.6.11 ??? :-S
But I try and grab different version or compile from source.
I will work around it if needed.

carlosperate, 2016-10-27:
So, if you've downloaded version 1.6.x, how is BlocklyDuino finding your local copy? Based on the log it falls down to the default arduino command. If you open the terminal and type `arduino` which version opens up?

talalz94, 2018-06-20:
@carlosperate 
Sorry to be a bother, I have a similar problem. When I upload the code, it returns the following error:

[image]

the log generated is:

C:\Users\bilal\Desktop\New folder\BlocklyDuino-gh-pages>arduino_web_server.py
Blocklyduino can now be accessed at http://127.0.0.1:8080/
127.0.0.1 - - [19/Jun/2018 15:27:35] "GET /blockly/apps/blocklyduino/index.html HTTP/1.1" 200 -
127.0.0.1 - - [19/Jun/2018 15:27:35] "GET /blockly/apps/blocklyduino/node.manifest HTTP/1.1" 200 -
sketch to upload: int i;

void setup()
{

}

void loop()
{
for (i = 1; i <= 10; i++) {
}

}
created sketch at c:\users\bilal\appdata\local\temp\tmpdlkiyx\tmpdlkiyx.ino
INFO:root:Couldn't find Arduino command; hoping it's on the path!
INFO:root:Guessing port name as COM8
Uploading with arduino --upload --port COM8 c:\users\bilal\appdata\local\temp\tmpdlkiyx\tmpdlkiyx.ino
Exception happened during processing of request from ('127.0.0.1', 51565)
Traceback (most recent call last):
File "C:\Python27\lib\SocketServer.py", line 295, in _handle_request_noblock
self.process_request(request, client_address)
File "C:\Python27\lib\SocketServer.py", line 321, in process_request
self.finish_request(request, client_address)
File "C:\Python27\lib\SocketServer.py", line 334, in finish_request
self.RequestHandlerClass(request, client_address, self)
File "C:\Python27\lib\SocketServer.py", line 655, in init
self.handle()
File "C:\Python27\lib\BaseHTTPServer.py", line 340, in handle
self.handle_one_request()
File "C:\Python27\lib\BaseHTTPServer.py", line 328, in handle_one_request
method()
File "C:\Users\bilal\Desktop\New folder\BlocklyDuino-gh-pages\arduino_web_server.py", line 136, in do_POST
rc = subprocess.call(compile_args)
File "C:\Python27\lib\subprocess.py", line 522, in call
return Popen(*popenargs, **kwargs).wait()
File "C:\Python27\lib\subprocess.py", line 710, in init
errread, errwrite)
File "C:\Python27\lib\subprocess.py", line 958, in _execute_child
startupinfo)
WindowsError: [Error 2] The system cannot find the file specified

shawt, 2018-06-20:
Is there a web server running locally to handle your code?

The web app will just produce the Arduino code. You need something running
on the computer to compile and upload the code to your board.

carlosperate, 2018-06-20:
Sounds like the python server is trying to find the Arduino installation, cannot find it, and crashes.
Have you install the Arduino IDE in the default location?

talalz94, 2018-06-22:
@shawt Isnt arduino_web_server.py creating the web server? 
What I am basically doing is that:

-I am using the app at 'BlocklyDuino-gh-pages/blockly/apps/blocklyduino/index.html' to create my code for arduino.

-I run arduino_web_server.py

-I click upload from the application (index.html) and I am presented with the error I just presented.

talalz94, 2018-06-22:
@shawt thanks for all the help. I got it the arduino IDE wasnt properly installed.

talalz94, 2018-06-22:
@carlosperate that did the trick. Thanks