BlocklyDuino/BlocklyDuino #69: transmit a value to a fonction

jbet, 2016-03-24:
Hello,
I need to use function, and to transmit two values, for robot and Arduino Uno.
With Blockly, no problem, we can do it.
The problem begin when we compile ;  It miss the type integer in front of the value.

Exemple  speed_right_motor must be transform by **int** speed_right_motor to have the right code :
 int speed_right_motor speed()

carlosperate, 2016-03-25:
All variables are globals on Blockly and BlocklyDuino, so if you have that value already set, you don't need any arguments in the function, you should  be able to just use it directly.

jbet, 2016-03-25:
Thank you for your answer.
I use Blocky@rduino.

I will test,and I give you an answer.

Where do you write, for me it's in France.

----- Mail original -----

016 09:40:09
Objet: Re: [BlocklyDuino] transmit a value to a fonction (#69)