BlocklyDuino/BlocklyDuino #70: Error transmit data to a fonction.

jbet, 2016-03-27:
When I compile this block

[image] 

I obtain this error
[image]

The error comme from the variable Vitesse_gauche, or Vitesse_droite, in the function Vitesse_robot, has no type.

Could you correct this bug, or is there a solution.

carlosperate, 2016-03-28:
There is a chance this could be an issue with your customised blocks rather than BlocklyDuino. Could you replicate using original BlocklyDuino blocks so that we could investigate?

Very hard to figure it out from a fork, specially without seeing what the code produced is. The default behaviour is to declare a void function if no return is specified, and all arguments are of type int.

jbet, 2016-03-28:
**I have do a simple example**
[image]

**Then the code for Arduino**
int led;
int Etat bouton 1;
int led 1;
void Allumer_Eteindre_LED1(led) {
  if (led == 1) {
    digitalWrite(1,HIGH);

  }
  if (led == 0) {
    digitalWrite(1,LOW);

  }
}

void setup() {
  pinMode(1, OUTPUT);
  pinMode((Etat_bouton_1 == true), INPUT);
  pinMode((Etat_bouton_1 == false), INPUT);
    Etat_bouton_1 = 1;
  led_1 = 2;

}

void loop() {
    if (digitalRead((Etat_bouton_1 == true))) {
      Allumer_Eteindre_LED1(1);

    }
    if (digitalRead((Etat_bouton_1 == false))) {
      Allumer_Eteindre_LED1(0);

    }

}

**Then then verification**
Verification failed. Erreur : (sketch file) sketch.ino:2:28: error: **unknown type name 'led' void** **Allumer_Eteindre_LED1(led);**                            ^
(sketch file) sketch.ino:2:9: error: expected ';' after top level declarator int Etat bouton 1;         ^
        ;
(sketch file) sketch.ino:3:5: error: redefinition of 'led' int led 1;     ^
(sketch file) sketch.ino:1:5: note: previous definition is here int led;     ^
(sketch file) sketch.ino:3:8: error: expected ';' after top level declarator int led 1;        ^
       ;
(sketch file) sketch.ino:4:6: error: variable has incomplete type 'void' void Allumer_Eteindre_LED1(led) {      ^
(sketch file) sketch.ino:4:32: error: expected ';' after top level declarator void Allumer_Eteindre_LED1(led) {                                ^
                               ;
6 errors generated.

**My analyse**
On this example, the problem come from that led, in function "Allumer_Eteindre_LED1(led)".
Led has no type, led is not an integer, it's a variable only for fucntion Allumer Eteindre ( in english switch on / off)

It's the first  bug of Blockly@rduino...
There is a second bug for ISR ( interrupt sub routine)

**Best regard**

carlosperate, 2016-03-28:
The following two lines are not valid C/C++ code:

```c
int Etat bouton 1;
int led 1;
```

And the function `Allumer_Eteindre_LED1` is not generating the function argument code correctly, as not type is set:

```c
void Allumer_Eteindre_LED1(led) {
  if (led == 1) {
    digitalWrite(1,HIGH);
  }
}
```

They both seem to be issues directly related with the changes made in Blockly@rduino rather than BlocklyDuino, so you'd probably be able to get better support in their github repository issue tracker. Maybe @SebCanet can  shed some light on the issue?

jbet, 2016-03-28:
Thank you for your answer. I will used Blocky@duino with pupils 12 year's to 15 year's old.
That the new " program" for technology.

Best regard.

closed 2016-03-28 by jbet