BlocklyDuino/BlocklyDuino #72: Traduction bluetooth block in french

jbet, 2016-03-30:
French  traduction
[image]
"Lire" is not the rignt traduction
Could you change the block  bluetooth, by remplacing "lire" by "envoyer" as we seen on the spanish traduction.

Spanish traduction
[image]

Best regard

carlosperate, 2016-03-30:
Are you sure you are using this BlocklyDuino? I believe there are no translation for the current blocks, and the Bluetooth block is a different colour. Apart from that the BlocklyDuiono version has "receive" first and then "send", which seems to be the opposite way displayed in your screenshot for both languages.

jbet, 2016-03-30:
I use this.
http://www.technologiescollege.fr/blockly@rduino/?lang=fr

carlosperate, 2016-03-30:
You will get better support in their repository then: https://github.com/technologiescollege/Blockly-at-rduino

gasolin, 2016-04-15:
Yeah BlocklyDuino does not provide translation yet

closed 2016-04-15 by gasolin