BlocklyDuino/BlocklyDuino #77: Help with if statement in generator script

jwill4, 2016-05-05:
I paste below my generator code for a Zumo bot motor block.  

It works fine if i drop the block directly on the blockly "stage", however if i try to drop this block inside another block (e.g. loop or conditional) it does not generate code.  It even seems to stop the loop or conditional that it is dropped into from generating code.

What am i doing wrong?

```
Blockly.Arduino['output_leftzmotor'] = function() {
  var dropdown_direction = this.getFieldValue('Direction');
  var value_speed = Blockly.Arduino.valueToCode(this, 'SPEED', Blockly.Arduino.ORDER_ATOMIC);

  Blockly.Arduino.definitions_['define_Zumo32U4Motors'] = '#include <Zumo32U4Motors.h>\n';
  Blockly.Arduino.definitions_['var_Zumo32U4Motors'] = 'Zumo32U4Motors motors;\n';

  if(dropdown_direction == "FORWARD"){
    var code = 'motors.setLeftSpeed('+value_speed+');\n';
  } else {
    var code = 'motors.setLeftSpeed(-'+value_speed+');\n';
  }

return [code, Blockly.Arduino.ORDER_ATOMIC];
};

```
Please tell me if this is the wrong place to ask for this help, or if i have other issues with my post.

jwill4, 2016-05-05:
Here is the block code, in case it is relevant.
```
Blockly.Blocks['output_leftzmotor'] = {
  init: function() {
    this.appendDummyInput()
        .appendField("Zumo Left Motor")
        .appendField(new Blockly.FieldDropdown([["Forward", "FORWARD"], ["Backward", "BACKWARD"]]), "Direction");
    this.appendValueInput("SPEED")
        .setCheck("Number")
        .appendField("Speed");
    this.setInputsInline(true);
    this.setPreviousStatement(true, null);
    this.setNextStatement(true, null);
    this.setColour(30);
    this.setTooltip('Set left motor speed and direction');
    this.setHelpUrl('http://www.example.com/');
  }
};
```

gasolin, 2016-05-06:
You should check with Zumo bot developer & report it in their repository for a customized block

gasolin, 2016-05-06:
Oops I saw you are the Zumo bot developer :p

Could you post the generated Arduino code? Eventually the source talks.

jwill4, 2016-05-06:
Thanks so much for the reply.
If i drop just a "Zumo Left Motor" block onto the Blocklyduino stage i get this:
```
#include <Zumo32U4Motors.h>

Zumo32U4Motors motors;

void setup()
{

}

void loop()
{
  motors.setLeftSpeed();

}
```
Which is perfect.  
(If i drop a number block into the speed input notch it is inserted correctly into the motors.setLeftSpeed() function call.  Again perfect).

However if i first drop a "repeat while do" block onto the stage and THEN try to insert the "Zumo Left Motor" into that "repeat while do" block i get this:
```
void setup()
{

}

void loop()
{
  while (false) {
  }

}
```

stevoleeto, 2016-05-06:
For the Blockly.Arduino generator code, just simply return code.

`Blockly.Arduino['output_leftzmotor'] = function() {
  var dropdown_direction = this.getFieldValue('Direction');
  var value_speed = Blockly.Arduino.valueToCode(this, 'SPEED', Blockly.Arduino.ORDER_ATOMIC);

  Blockly.Arduino.definitions_['define_Zumo32U4Motors'] = '#include <Zumo32U4Motors.h>\n';
  Blockly.Arduino.definitions_['var_Zumo32U4Motors'] = 'Zumo32U4Motors motors;\n';

  if(dropdown_direction == "FORWARD"){
    var code = 'motors.setLeftSpeed('+value_speed+');\n';
  } else {
    var code = 'motors.setLeftSpeed(-'+value_speed+');\n';
  }

return [code, Blockly.Arduino.ORDER_ATOMIC];
};`

I just tested it and it should work.

jwill4, 2016-05-06:
Thank you @stevoleeto for your response.
I don't understand.  Wasn't i returning code ?
I cant see the difference between my code and yours, except in the following line:
`Blockly.Arduino.definitions_['define_Zumo32U4Motors'] = '#include <Zumo32U4Motors.h>\n';`
where you have:
`Blockly.Arduino.definitions_['define_Zumo32U4Motors'] = '#include \n';`
Regardless i copied and pasted your code into the generator file, but found no change in its operation.

jwill4, 2016-05-06:
Here's some more strange behaviour of the same motor block.

If i place the motor block BEFORE or AFTER another block the code generated is appended as would be expected but with a strange artefact (,0).  The code below was generated when i first placed the delay block then attached the motor block to the end of it.
```
#include Zumo32U4Motors motors;

void setup()
{

}

void loop()
{
  delay(1000);
  motors.setLeftSpeed();
  ,0
}
```
Any ideas?

jwill4, 2016-05-06:
Genius!  Thank you, @steveleeto, so simple.  
I thought i had tried that.  I am rushing because i am assisting with a Code Camp for 12-14 year olds this weekend. 
THANK YOU SO MUCH FOR YOUR TIME.
Sorry @stevoleeto i should have understood what you meant the first time.  

I have updated my Zumo32U4 repository.  Thank you.

closed 2016-05-06 by jwill4

stevoleeto, 2016-05-07:
Glad you got it working, and good luck with your code camp!