BlocklyDuino/BlocklyDuino #89: Creating new blocks from existing blocks

KristofBostoen, 2017-02-07:
Dear ArduBlockly wizards,

I'm looking for a simple way of getting 8-14 years to code and would like to do this by getting them to code an Arduino based robot. I would like to simplify the code for these kids and write some predefined blocks to simplify things. I looked around but I could not find a simple way to do this and was wondering if I'm missing something. I just would like to create a folder with these blocks and ensure that it is visible in the IDE.

Looking forward to any help.
Kind regards
Kristof

shawt, 2017-02-07:
Hi Kristof.

As you probably know, in blockly, there are two parts that affect how a block works:

- Block Definition Code: saved in one of the files in blockly/blocs
- Code Generator Stub: saved in one (or more) of the files in blockly/generators

If you have a block that is close but not exactly doing what you want, you can copy the function that defines a block, change the name and paste it right below. You then do the same thing for the generator stub function that corresponds to that block. Be sure to change the name on your generator code too, so that it matches the name in the definition.  Once you have your functions set, you will need to run build.py again.

If you want to build a block from scratch, there is an arduino version of the Block factory here: http://blocklyduino.github.io/BlocklyDuinoFactory/  This will help you generate the definition and generator stubs from scratch.

You may also want to take a look at my tool here: http://iforge.genlrn.com .

Good luck!

KristofBostoen, 2017-02-08:
Thanks a lot for this.
I was a bit confused (given I posted on the wrong forum)  as I'm using
ArduBlockly rather than BlocklyDuino and wondering how similar or different
they are given that there seems to be a crossover of developers.
Will give it a try and let you know how successful I am or if I need a more
specific helping hand :-).

Kind regards
Kristof

closed 2017-03-08 by gasolin