BlocklyDuino/BlocklyDuino #96: Please help me to make a new blocks <3 <3 <3

hamam1, 2017-09-29:
Please help me to make a new blocks 
i try many times but i can't make a new block

shawt, 2017-09-29:
I can probably help. What exactly are you trying to do, and where are you
getting stuck?

Creating new blocks includes three basic steps...

1.) Define the shape and look of your block using the block factory tool
<http://blocklyduino.github.io/BlocklyDuinoFactory/>.

2.) This will generate two chunks of code that need to be moved into your
source files:

the definition code, which gets pasted as-is into one of the files in your
/blocks directory. It doesn't matter what file. Just put it one that makes
sense to you.

The generator code for arduino which goes into a file in the /generators
directory.  This code needs to be edited so the block knows what code you
want it to produce.  You have to create a value for the code variable which
will be returned by the function.

3.) Then run build.py from your project root directory to create the
compressed files

Don't forget to add your newly created block to your toolbox.xml file.

Trevor

-- 
**************
Trevor Shaw
President
Genesis Learning LLC
(v) 866.797.0398
@gen_learn
shawt@genlrn.com
http://www.genlrn.com
*****************

hamam1, 2017-09-29:
i made that a many of time
can you make vedio please <3 <3 <3 <3 <3

hamam1, 2017-10-03:
pleas any one help me  <3 <3 <3

talalz94, 2018-06-14:
@shawt I followed your tutorial but when I ran the build.py, it gave the following error:

Error: Closure not found.  Read this:
https://developers.google.com/blockly/hacking/closure

-I only made changes to the "C:\Users\bilal\Desktop\BlocklyDuino-gh-pages\blockly\blocks\grove.js", where I edited the block slightly just as a test

shawt, 2018-06-15:
Hi Talal,

Did you read the link and verify that the closure library is located in the
right spot.

I sometimes get this error because I usually leave the closure library out
of my git repo, so when I work on a different machine and clone my repo, I
forgot to download the closure library.

Trevor

-- 
**************
Trevor Shaw
President
Genesis Learning LLC
(v) 866.797.0398
@gen_learn
shawt@genlrn.com
http://www.genlrn.com
http://iforge.genlrn.com (block-based Arduino)
*****************

talalz94, 2018-06-19:
@shawt Thanks. That did the trick. The program now compiles successfully. 

Another issue though. After I run the build.py, the index.html stops working.

[image]

As you can see, I cant click on the blocks arudino or the xml tabs. Any idea what might be happening?

shawt, 2018-06-19:
I bet it's a syntax error in one of your block definitions or generators.

talalz94, 2018-06-19:
@shawt It turned out Microsoft Edge was causing the problem.

Sorry to be a bother, a final problem. When I upload the code, it returns the following error:

[image]

the log generated is:

C:\Users\bilal\Desktop\New folder\BlocklyDuino-gh-pages>arduino_web_server.py
Blocklyduino can now be accessed at http://127.0.0.1:8080/
127.0.0.1 - - [19/Jun/2018 15:27:35] "GET /blockly/apps/blocklyduino/index.html HTTP/1.1" 200 -
127.0.0.1 - - [19/Jun/2018 15:27:35] "GET /blockly/apps/blocklyduino/node.manifest HTTP/1.1" 200 -
sketch to upload: int i;

void setup()
{

}

void loop()
{
  for (i = 1; i <= 10; i++) {
  }

}
created sketch at c:\users\bilal\appdata\local\temp\tmpdlkiyx\tmpdlkiyx.ino
INFO:root:Couldn't find Arduino command; hoping it's on the path!
INFO:root:Guessing port name as COM8
Uploading with arduino --upload --port COM8 c:\users\bilal\appdata\local\temp\tmpdlkiyx\tmpdlkiyx.ino
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 51565)
Traceback (most recent call last):
  File "C:\Python27\lib\SocketServer.py", line 295, in _handle_request_noblock
    self.process_request(request, client_address)
  File "C:\Python27\lib\SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "C:\Python27\lib\SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Python27\lib\SocketServer.py", line 655, in __init__
    self.handle()
  File "C:\Python27\lib\BaseHTTPServer.py", line 340, in handle
    self.handle_one_request()
  File "C:\Python27\lib\BaseHTTPServer.py", line 328, in handle_one_request
    method()
  File "C:\Users\bilal\Desktop\New folder\BlocklyDuino-gh-pages\arduino_web_server.py", line 136, in do_POST
    rc = subprocess.call(compile_args)
  File "C:\Python27\lib\subprocess.py", line 522, in call
    return Popen(*popenargs, **kwargs).wait()
  File "C:\Python27\lib\subprocess.py", line 710, in __init__
    errread, errwrite)
  File "C:\Python27\lib\subprocess.py", line 958, in _execute_child
    startupinfo)
WindowsError: [Error 2] The system cannot find the file specified

shawt, 2018-06-19:
If you are running off of the demo app, I think the idea is that the button
pushes the code to a web server running locally that will compile and flash
the code to a board the way that Arduino Create <https://create.arduino.cc>
does.

This is not part of the sample program, you would have to create this
yourself.  The Arduino Create Agent
<https://github.com/arduino/arduino-create-agent>is open source, so you
could modify it for your own purposes if you wanted to.

Trevor

-- 
**************
Trevor Shaw
President
Genesis Learning LLC
(v) 866.797.0398
@gen_learn
shawt@genlrn.com
http://www.genlrn.com
http://iforge.genlrn.com (block-based Arduino)
*****************

mesut25, 2018-12-31:
http://blocklyduino.github.io/BlocklyDuinoFactory/

oluşturdugum kodu    göremiyorum  basit bir  video anlatım var mı?  .xml     oluşturdugum konu nasıl görebilirim.....