BlocklyDuino/BlocklyDuino #104: Fork from Blockly

SebCanet, 2018-03-13:
Hi,
I'm working on my Blockly@rduino, and I thought about forking...
As my project was a new project, with some of your code, I created new repo without forking yours. But maybe for easy update between project we should create a new BlocklyDuino, forked from Blockly which permit to benefit of all new update from them (numerous update...).
After that I'll re-create mine by forking yours.

What do you think about that ?
Regards.

gasolin, 2018-03-14:
Yeah I'd like to have update-to-date BlocklyDuino based on Blockly as well. Is Blockly@rduino add any new blocks or new features based on BlocklyDuino?

SebCanet, 2018-03-14:
I add a lot of feature, some thanks to @carlosperate , some are just visual enhancement.
What I was thinking was just a BlocklyDuino, as simple as possible, just something over Blockly. That's why I suggest a new BlocklyDuino repo based upon a fork from Blockly.
I tested the latest Blockly, put your egenrato file on it and it seems to work perfectly.

gasolin, 2018-03-14:
@SebCanet great to hear it works with latest Blockly!

I created branch `v2` https://github.com/BlocklyDuino/BlocklyDuino/tree/v2 as PR destination. You can send pull request to `v2` and we can see what we can improve from the new base.

It will be nice if we can make all BlocklyDuino related changes to a separate folder and patch it into Blockly. But it's not necessary for the first attempt.

SebCanet, 2018-03-14:
I started something from Blockly repo : [https://github.com/BlocklyDuino/BlocklyDuinoReboot](https://github.com/BlocklyDuino/BlocklyDuinoReboot)

What do you think of this way ?

gasolin, 2018-03-16:
All changes in https://github.com/BlocklyDuino/BlocklyDuinoReboot/commit/badc5d8395d8823b8119718144310b5762c2ea3a ?

I like put all changes in BlocklyDuino folder :+1: 
Do you think if we could put Blockly as a submodule? like [node-blockly](https://github.com/mo4islona/node-blockly) did https://github.com/mo4islona/node-blockly/blob/master/package.json#L26

As I said welcome to send the PR to https://github.com/BlocklyDuino/BlocklyDuino/tree/v2

SebCanet, 2019-01-07:
That's what I started to do with the BlocklyDuino Reboot, and re write the generator to make compatible with the new Bockly. But I still have some few pb with it...

philippejadin, 2019-01-07:
Yes but you need to separate google blockly and blocklyduino in two separate directories and never touch anything in google blockly else you cannot update. I started working on it here : https://github.com/philippejadin/BlocklyDuino/tree/v2

Basically I'm taking latest google blockly and putting back the bits you added in reboot, but I force myself to only work in /blocklyduino :-)

When it will  work, I'll clean everything and propose a pull request :-)

SebCanet, 2019-01-07:
Ok, I see what you mean, no pb I do not know Github thi much to be able to decide the good method, perfect if you know.
Thanks, it will replace actual BlocklyDuino Reboot.

philippejadin, 2019-01-07:
And here is the pull request : https://github.com/BlocklyDuino/BlocklyDuino/pull/108

You can also test it with : `git clone --recurse-submodules https://github.com/philippejadin/BlocklyDuino`
Then open /blocklyduino/index.html

The sooner we have a common ground, the better, so if you like it, accept the pr :-)

closed 2020-05-18 by SebCanet

philippejadin, 2020-05-18:
By the way, submodules might create problems as well, so let me know how it goes with this "solution"

SebCanet, 2020-05-18:
Did you have a look on v2? What do you think about it?
Is it what you proposed?