BlocklyDuino/BlocklyDuino #106: Using images in block

logix5, 2018-07-21:
Hi !! 

We are updating blockyDuino for my board and i am adding more sensors, ... At the moment is ok, and i am working in offline mode. I have shared the project with more people to test  the block, i have had a little surprise withe the images of my blocks.

I like to use images because the student prefer, but when other colleagues use the version in other operating system as linux, not all images appear and i don´t understand the reason. I though that all images would be show or not , but in some cases only a fews.
This is my repository update the Adel version.
https://github.com/logix5/Blockyduino-para-MRTDuino

[Blocklyduinocap.zip](https://github.com/BlocklyDuino/BlocklyDuino/files/2216102/Blocklyduinocap.zip)

I have attached examples of this behaviour. do you know what is the reason?

Thanks in advance
BR
Óscar

SebCanet, 2018-07-21:
Hi,
your project is cool!
What is your MRT Duino board ? Where can I have information about that ?
Quickly answering, I have this kind of problem when I work on Windows because it is not upper and lower case sensitive.
Could you verify ?

logix5, 2018-07-21:
Hi 
Thanks!! you are right because this files the extension was .PNG. I have updated tp .png and in my computer there is no problem but i will test on Monday in my laptop with this issue. I let you know.

i am working hard and i want to add more and more block with different sensors ans actuator (mp3, sd, serveral displays, ... all of them). At the moment i have create this block in Scratch using extension but i want to use a more complete and powerful blocks. 

This kit is:
https://www.logix5.com/roboticaeducativa/kit-beduino/
https://www.logix5.com/roboticaeducativa/producto/beduino/

Information about the board and examples and extension for Scratch is in these repositories
https://www.logix5.com/roboticaeducativa/producto/beduino/ and
https://github.com/logix5/Programas_MRTduino

Tha board is an Arduino Leonardo with motor driver. The pin out is in the file Pin-out and Board description.xlsx in the Programas_MRTduino repository.

The information is in spanish, we will migrate to english in the future. Blockyduino is creating now firstable in english. 

If you want more information i let you know. I have a company with other colleague about robotic course, and i am a teacher too, and i want the best tool to use with my student. We have created beduino kit using the pieces of My Robot Time company. Is our creation for our course, firstable use their electronic components and board and in the next course we are adding more arduino compatible sensors and actuators.

Thanks for all
Óscar

closed 2018-10-04 by logix5