BlocklyDuino/BlocklyDuino #109: use arduino-cli instead of Arduino IDE

gasolin, 2019-01-12:
use https://github.com/arduino/arduino-cli instead of install full Arduino IDE

or with js npm wrapper https://www.npmjs.com/package/arduino-cli

philippejadin, 2019-01-12:
I'm working on it here : https://github.com/philippejadin/arduino-cli-agent
It's a small script that run in the terminal and provides a server for compilation and upload.

There is both a python and a nodejs version. But both are work in progress. Would you prefer to use python or nodejs? I'm not entirely settled on this. It seems that with nodejs, making crossplatform binary build is easier, and this open the path to electron fat client. I say fat because, you quickly endup with a 150 mb binary...

SebCanet, 2019-01-12:
Just my point of vue...
For modularity I prefer Python script, this way everybody can embed it in C soft, or anything else.
Nodejs is for...nodejs only ? And as you said it's very FAT. And a lot of people in education use slow/old computer.
But if you can do two version, that's great :-D
Thanks for all  your work, you relaunched BlocklyDUino, ended my never finished BlocklyDuino Reboot, and we would be able to update Blockly thanks to new generator you debugged.

philippejadin, 2019-02-01:
Maybe as a first step I could just adapt the current python server of blocklyduino to use arduino-cli ?
As I understand, it "just" need to respond to a post request to / with the code provided there.

This would be a nice small first step. If you  agree it's a good idea, I'll work on it and make a pull request.

gasolin, 2019-02-02:
sure, any progress is appreciated :)

SebCanet, 2020-05-18:
Added in v2.

closed 2020-05-18 by SebCanet