BlocklyDuino/BlocklyDuino #113: can't find the given path as Step 4

kfkafka, 2020-03-31:
[image]
as mentioned in step 4. i cant find the folder 'apps'. Please help.

SebCanet, 2020-04-01:
Hi,
did you check you have the same folders as in the repository?
When you unzip, you should have folders like this : "BlocklyDuino-master/blockly/apps/blocklyduino/"

kfkafka, 2020-04-04:
[image]
Hi,
yes, I have checked, there is no folder in the repository and the unzipped folder

SebCanet, 2020-04-04:
Hi,
it seems you have too many blockly installed in the same folder :-D
When you unzip BlocklyDuino, you have folders like this screen capture:
[image]

In blue it's all the folders, in red it's what you have to open to find 'index.html'.

kfkafka, 2020-04-09:
Okay, so would please share the link to the right BlocklyDuino which would have the exact files

SebCanet, 2020-04-09:
You could download : [https://github.com/BlocklyDuino/BlocklyDuino/archive/gh-pages.zip](https://github.com/BlocklyDuino/BlocklyDuino/archive/gh-pages.zip)
Or tes new version: [https://github.com/BlocklyDuino/BlocklyDuino-v2/archive/master.zip](https://github.com/BlocklyDuino/BlocklyDuino-v2/archive/master.zip)

closed 2020-05-18 by SebCanet