BlocklyDuino/BlocklyDuino #117: Something goes wrong in index.html after build.py

skill91002, 2020-11-06:
I tried to build BlocklyDuino on my local device, but encountered some problems,
can anyone help me?

p.s. I clone the project, and build it directly. Nothing changes in code.

Before I run "python build.py" in BlocklyDuino/blockly,
everything works fine
[image]

But After I build it, the blocks on the webpage disappeared,
I can't push any buttons now.
[image]

 I think the new blockly_compressed.js caused this problem.

But I don't know how to fix it, since there's no error message when compiling it.

Thank you!

skill91002, 2020-11-06:
I know what's going on.
It's because some functions are missing in blockly_compressed.js.
In my way, I add these definitions in it, and the webpage works fine.

_goog.isString=function(a){return"string"==typeof a};
goog.isNumber=function(a){return"number"==typeof a};
goog.isFunction=function(a){return"function"==goog.typeOf(a)};
goog.isArray=function(a){return"array"==goog.typeOf(a)};_

closed 2020-11-06 by skill91002

SebCanet, 2020-11-06:
Hi,
this project is not maintained anymore, you should give atry at new version BlocklyDuino2: [https://github.com/BlocklyDuino/BlocklyDuino-v2](https://github.com/BlocklyDuino/BlocklyDuino-v2)
I've not yet finished code generation for all blocks, but everything else works fine.